odinb/bash-plexamp-installer #1: 2.0 beta not working anymore

mcr-ksh, 2022-02-03:
`error: DEVICE: Error loading media providers from plex.tv: Error: Request failed with status code 406
error: DEVICE: Error loading providers or resources, not merging.
warn: DEVICE: Did not find a working server.
`

also using the 1.1.0 client on mac no libraries are found. I guess this is it to the old install.

odinb, 2022-02-03:
Hi! Currently this script is broken due to updates of how the authentication happens with the plex-server, and it does not look like that will be fixed, see more here:
https://forums.plex.tv/t/plexamp-raspberry-pi-certificate-expired/747294

But looks like a new version of this client is in the works:
https://forums.plex.tv/t/future-of-plexamp-on-raspberry-pi/717577

mcr-ksh, 2022-02-03:
appreciate the fast response. This is not due to the DST root certificate. I have no error like that. Rather the `Error loading media providers from plex.tv` which is status 406 = not acceptable. some API error. Does it work for you atm?

closed 2022-05-27 by odinb