odinb/bash-plexamp-installer #2: `-bash: Post: command not found`

djdembeck, 2022-06-10:
Seems echo isn't commented correctly:
https://github.com/odinb/bash-plexamp-installer/blob/main/install_Plexamp_pi.sh#L393

odinb, 2022-06-10:
Looks Ok when I run it:
--== For Linux 5.4 and higher ==--
 This not needed for the PiFi HIFI DiGi+ Digital Sound Card found at:
      https://www.fasttech.com/p/5137000

      If correct overlay is configured, but the system still doesn’t load the driver,
      disable the onboard EEPROM by adding: 'force_eeprom_read=0' to '/boot/config.txt'

--== End of Post-PlexAmp-script, please reboot for all changes to take effect ==--

 Configuration post-reboot:
      Note !! Only needed if fresh install, not if upgrading. Tokens are preserved during upgrade.
      After reboot, as your regular user please run the command: node /home/brodersen/plexamp/js/index.js
      At this point, go to the URL provided in response, and enter the claim token at prompt.

      Once entered, the web-GUI should be available on the ip-of-plexamp-pi:32500 from a browser.
      On that GUI you will be asked to login to your Plex-acoount for security-reasons,
      and then choose a librabry where to fetch/stream music from.
      Now play some music! Or control it from any other instance of Plexamp.

      Start and enable the Plexamp service if you feel like having it start on boot!
      Hit ctrl+c to stop process, then enter:
      systemctl --user enable plexamp.service && systemctl --user start plexamp.service

      NOTE!! If you upgraded, only reboot is needed, tokens are preserved.
      One can verify service after reboot with: systemctl --user status plexamp.service
      All should work at this point.

      Logs are located at: ~/.cache/Plexamp/log/Plexamp.log

What OS are you using? What does it look like on yours?

odinb, 2022-06-10:
Added quotes, see if that helps.

closed 2022-06-10 by odinb