odinb/bash-plexamp-installer #5: Status rpi-eeprom-update hangs

pscharfg, 2022-08-04:
I am not sure why this is happening but when your script reaches 'systemctl status rpi-eeprom-update.service’, the script hangs because it is not returning to the command prompt.  I am using a RPI4 and the status shows all is well, it just doesn't return.

To get around this in the past, I simply commented out the line in your script.  However changing it to `systemctl status rpi-eeprom-update.service --no-pager -l` is a better solution.

I assume this is not happening on your device and I can't figure out why mine would behave differently.  Any thoughts?

odinb, 2022-08-04:
Yup, fixed! Thanks for reporting!

closed 2022-08-04 by odinb

pscharfg, 2022-08-04:
No problem.  I didn't report it earlier because I thought there must be something wrong with my config.

odinb, 2022-08-04:
Not necessarily, there apparently has been a fairly recent change to systemctl, and then depending on your profile-settings, this might or might not happen! I did not see this when I ran the script personally, and most other people did not either, or I would have gotten more complaints (LOL)! But making the script more robust is always a goal, and this does just that!