odinb/bash-plexamp-installer #6: Installer doesn't create systemctl files on DietPi

dlasher, 2022-08-10:
First, thank you for creating this installer, fantastic work!

Second, I have an army of dietpi machines, and I ran the installer on a couple tonight. Plexamp can be started by hand (node js/index.js) and when you do, it works just fine. However, the installer didn't create any of the systemD files needed to set it to start on boot.

I'll go digging through the script, but hoping you can provide them as stand-alone in the short term, and add support tin the installer as a fix?

thank you!

odinb, 2022-08-10:
Will have a look. Might take a couple of days, as I am quite busy with other things, and DietPi is best effort, since I do not run it myself.

dlasher, 2022-08-10:
Fair enough, thank you @odinb for the quick response. 

based on getting it running, it appears to need a couple of things:
1. correct user/group/working directory
2. enable as root, rather than --user

Here's what I ended up with:

[Unit]
Description=Plexamp
After=network-online.target
Requires=network-online.target

[Service]
Type=simple
User=dietpi
Group=dietpi
WorkingDirectory=/home/dietpi/plexamp
ExecStart=/usr/bin/node /home/dietpi/plexamp/js/index.js
Restart=on-failure

[Install]
WantedBy=basic.target

odinb, 2022-08-14:
Hi! Should work now, please remove your manually installed/configured unit-file for the service, and re-run the installer!
Let me know any findings, and I will close once I see your response.

closed 2022-08-19 by odinb