odinb/bash-plexamp-installer #7: Small issues when using installation script

EgonHeuson, 2022-09-17:
First of all, thanks a lot for this script! It's amazing, and when I'll make it work properly I'm sure it will be perfect. I tried the installation on a RPi 4 without a DAC (just to try). I get few issues that I list below:

When starting Plexamp (4.3.0), I get this: 
`Starting Plexamp 4.3.0
DEVICE: No provider for source 80df67e4336a1a1c9911d2b4b6b4133c9f7d4cab`
Any idea what this is about? 

I also got an error during the first Plexamp install and start up:
```
xxx@PlexampPi:~ $ node /home/xxx/plexamp/js/index.js
Starting Plexamp 4.3.0
Please visit https://plex.tv/claim and enter the claim token: claim-xxxxxxxxxxxxxxxxxxxx
Please give the player a name (e.g. Bedroom, Kitchen): Marantz
DEVICE: Error loading cloud players from plex.tv HTTP status 403
```

Also, when I execute the following commands I get: 
```
xxx@PlexampPi:~ $ systemctl --user enable plexamp.service && systemctl --user start plexamp.service
Created symlink /home/xxx/.config/systemd/user/basic.target.wants/plexamp.service → /home/xxx/.config/systemd/user/plexamp.service.
Failed to start plexamp.service: Unit network-online.target not found.
```

Same for: 
```
xxx@PlexampPi:~ $ systemctl --user restart plexamp.service
Failed to restart plexamp.service: Unit network-online.target not found.
```

But what's stranger, is that the Plexamp web interface seems to work. Of course I had the issues details at the bottom of the Github tutorial (white wheel spinning forever), but after selecting the correct audio output source, everything's ok.
My only complain is that the sound coming from the jack is just trash (practically inaudible) and I do not know if it comes from the hardware or my config... Usually the jack output is not that bad and I can averagely listen to some music with it. Any suggestion?

odinb, 2022-09-17:
===
Q:
When starting Plexamp (4.3.0), I get this:
Starting Plexamp 4.3.0 DEVICE: No provider for source 80df67e4336a1a1c9911d2b4b6b4133c9f7d4cab
Any idea what this is about?

A:
You need to choose your source, i.e. Plex server or TIDAL other. This can be done from the GUI.

===
Q:
I also got an error during the first Plexamp install and start up:

xxx@PlexampPi:~ $ node /home/xxx/plexamp/js/index.js
Starting Plexamp 4.3.0
Please visit https://plex.tv/claim and enter the claim token: claim-xxxxxxxxxxxxxxxxxxxx
Please give the player a name (e.g. Bedroom, Kitchen): Marantz
DEVICE: Error loading cloud players from plex.tv HTTP status 403

A:
DEVICE: Error loading cloud players from plex.tv HTTP status 403

That is not a bad thing per se, it just means you didn’t link any “cloud players”.

See also:
https://forums.plex.tv/t/plexamp-headless-not-playing-music/809078/21?u=odinb1

===
Q:
Also, when I execute the following commands I get:

xxx@PlexampPi:~ $ systemctl --user enable plexamp.service && systemctl --user start plexamp.service
Created symlink /home/xxx/.config/systemd/user/basic.target.wants/plexamp.service → /home/xxx/.config/systemd/user/plexamp.service.
Failed to start plexamp.service: Unit network-online.target not found.

Same for:

xxx@PlexampPi:~ $ systemctl --user restart plexamp.service
Failed to restart plexamp.service: Unit network-online.target not found.

A:
Sounds like you did not reboot the RPi! Try rebooting and try again!
The service-file is from Pleaxmp, I just modify it for the user. No modifications to the "[Unit]" part.

closed 2022-09-17 by odinb

EgonHeuson, 2022-09-18:
Thanks for your answers! Ok for the 2 first questions :-)
About:
> Sounds like you did not reboot the RPi! Try rebooting and try again!
> The service-file is from Plexamp, I just modify it for the user. No modifications to the "[Unit]" part.
I did reboot the Pi right after the installation, and actually rebooted it several times since but still get the issue...

odinb, 2022-09-18:
What does that service tell you? Mine shows:
root@plexamp4:~# systemctl status network-online.target
● network-online.target - Network is Online
     Loaded: loaded (/lib/systemd/system/network-online.target; static)
     Active: active since Sun 2022-09-18 15:04:03 CDT; 2min 25s ago
       Docs: man:systemd.special(7)
             https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget

Sep 18 15:04:03 plexamp4 systemd[1]: Reached target Network is Online.
root@plexamp4:~#

Also, the service should restart if you reboot. As stated, I did not create the service-file, I only modify it for the user.

EgonHeuson, 2022-09-22:
Hello! I add up to this thread. I received my Pifi DAC+ and tried a new install. Everything seemed to work flawlessly excepted the last part. If I run `sudo systemctl status plexamp.service` after a reboot, I get the status correctly working as seen below:

[image]

However, no way to access the GUI over the web browser. To have access to it, I need to manually run Plexamp `node /home/"$USER"/plexamp/js/index.js`. Do you know if it's a normal behavior? I'm pretty sure it doesn't come from the installation script, but I wondered if your already saw this? 

Also, few other remarks. When executing the following command `sudo systemctl --user status plexamp.service` as _pi_ or as _root_, I always get: 
`Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)`

However the command works perfectly if I remove the `--user` argument. Can you let me know why you added the latter?

EgonHeuson, 2022-09-22:
Adding up to this, I was playing around to understand better what could happen, and at some point I checked if plexamp was correctly installed in the user I defined. I `cd` into its home folder and everything seemed fine. I re-ran the command `node /home/"$USER"/plexamp/js/index.js` from there and Oh Surprise, it asked me again to claim a token while I already performed this action right after the first reboot after the installation. And now it seems to be working correctly, even after reboot, with the web interface accessible without having me to execute again de manual command `node /home/"$USER"/plexamp/js/index.js`.

But (because there is always a "but") it seems that at reboot, the output is automatically changed for my HDMI (instead of my HAT):

And I get this:

`xxx@RPiPlexamp:~ $ sudo systemctl status plexamp.service
● plexamp.service - Plexamp
     Loaded: loaded (/home/xxx/plexamp/plexamp.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-09-22 10:49:17 CEST; 9min ago
   Main PID: 648 (node)
      Tasks: 27 (limit: 1830)
        CPU: 7.243s
     CGroup: /system.slice/plexamp.service
             └─648 /usr/bin/node /home/xxx/plexamp/js/index.js

Sep 22 10:49:17 RPiPlexamp systemd[1]: Started Plexamp.
Sep 22 10:49:18 RPiPlexamp node[648]: Starting Plexamp 4.4.0
Sep 22 10:49:18 RPiPlexamp node[648]: DEVICE: No provider for source 80df67e4336a1a1c9911d2b4b6b4133c9f7d4cab
Sep 22 10:49:19 RPiPlexamp node[648]: ALSA lib pcm_dmix.c:1075:(snd_pcm_dmix_open) unable to open slave
Sep 22 10:49:19 RPiPlexamp node[648]: ALSA lib pcm_dmix.c:1075:(snd_pcm_dmix_open) unable to open slave
Sep 22 10:49:19 RPiPlexamp node[648]: ALSA lib pcm_dmix.c:1075:(snd_pcm_dmix_open) unable to open slave`

odinb, 2022-09-22:
Hi!
No need to run the --user anymore, I changed the service from a user-service to a system service, but still start the process as the user.This since as a user-service, it did not have access to if networking was up, as a system-servie it does!
So, just run "systemctl status plexamp.service" with or without the sudo!
You should not have to run the "/home/"$USER"/plexamp/js/index.js" more than on initial setup, once done, it should not be needed, not even after upgrades.
The ALSA-error you are seeing is addressed in the Q&A on the gihub wiki, you need to re-choose your DIGI+ in the GUI!If not visible in the GUI, try a reboot and then try choosing it again!
Hope this helps!
Regards,
//Odin

EgonHeuson, 2022-09-22:
> No need to run the --user anymore, I changed the service from a user-service to a system service, but still start the process as the user.
Ok good to know. Still behavior is quite strange. Sometimes it boots up and Plexamp starts perfectly without any error, and sometimes I have the Alsa error. If I reboot then it works fine again... Very strange... I'll try a manual install someday to understand better.

odinb, 2022-09-22:
Just pick the correct audio card in the webGUI, and the ALSA-error will go away!

EgonHeuson, 2022-10-08:
Hey! I'm back with some news. I finally succeeded in making my Pfi DAC+ working very fine with headless Plexamp. In my case it, I have less issue with manual Plexamp installation, before running the script. When I use to use the script directly, I needed to start manually Plexamp most of the time, but with a manual install as reported here, I have no more this issue. 

However, as you advised, I ordered, and just received, a Pifi Digi HAT to directly use my Marantz DAC. However, when using on my Raspberry Pi 4 that I received in May-June, I cannot make it appearing in the available sound cards. I tried bot manual procedure (from Hifiberry website) and using your script for its installation. But I cannot see it. I attach some info:

Here it is what I get with command `aplay -l` and I performed a manual install using Hifiberry instructions
`root@RPiPlexamp:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`
Only the HDMI 1 is listed. 

And when I use your script:
`root@RPiPlexamp:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

The 2 HDMI are listed, but no HAT detected, also in Plexamp interface...

[image]
 So I wondered if maybe this card is not compatible anymore with the latest RPi4 version. I read that Hifiberry Digi2 Pro was not compatible in version 2.1 with latest versions of the RPi4.

Any idea?

odinb, 2022-10-09:
Hi!

Looks like the card is not properly configured in Pi OS, I get:
`root@plexamp4:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@plexamp4:~#`

Which card exactly did you buy? URL and config instructions for it, please!

EgonHeuson, 2022-10-09:
I bought this one: [https://fr.aliexpress.com/item/32998739264.html?spm=a2g0o.order_list.0.0.40d45e5bzPu8Vq&gatewayAdapt=glo2fra](https://fr.aliexpress.com/item/32998739264.html?spm=a2g0o.order_list.0.0.40d45e5bzPu8Vq&gatewayAdapt=glo2fra)
Actually I think it's the one from the link you shared with me in your advice on the [Plex Forum](https://forums.plex.tv/t/issue-with-plexamp-and-airplay-music-streaming-skipping-songs-and-extreme-latency/810958/11). 

Regarding the config instruction, I went on the [Hifiberry website](https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/), and used the info for the Hifiberry Digi+. I also tried with your script. And I also found this [page](https://www.nielsmayer.com/xwiki/bin/view/Raspberry+Pi/Kodi+Audio+Heaven+With+PiFi+Digi+SPDIF+HAT+and+External+DAC) but it is quite outdated. Tried some attempts following it but didn't help much yet. 

Regarding the Raspberry Pi OS, I run the latest OS from their website, in lite 64bits version:
[image]

> Also, what do you have at the end of /boot/config.txt?
> 
> You should have something like this:
> # --== Configuration for HiFi-Berry ==-- dtoverlay=hifiberry-digi

I do have it. I also tried to add it manually following the Hifiberry procedure. 
I will send it here. i'm away for the night, I come back home tomorrow evening. :-)

odinb, 2022-10-10:
Well, then I do not know, because my script should take care of it! I have 4 of these cards on 4 different RPi's, and they all work with my script after fresh install! I have hundreds using my script, and it seems to work well for them! I do however run the "Raspberry Pi OS Lite", not the desktop version. Headless means no GUI! Have you tried that? If it still does not work on the lite, maybe you have a bad card?

EgonHeuson, 2022-10-10:
Ok :-) I thought that maybe, as for the last Hifiberry Digi2 Pro, some versions of the card might only be compatible with the previous version of the RPi4
P.S. As I show in the previous picture, I also use the "Raspberry Pi OS Lite". So the problem doesn't come from this. Unless you use the 32 bits version maybe?

odinb, 2022-10-10:
Nope, 64-bit all the way! And you do not have the Digi2 Pro, you have the Digi+ clone, so not applicable! But if you did have it, the workaround is this:
If you have configured the correct overlay, but the system still doesn’t load the driver, this might be caused by these changes. In this case, you need to disable the onboard EEPROM by adding
'force_eeprom_read=0'
to your /boot/config.txt file.

Listed some output from mine that is working:
'root@plexamp4:~# cat /boot/config.txt |grep dtparam=audio=
#dtparam=audio=on
root@plexamp4:~# cat /boot/config.txt |grep dtoverlay=vc4-
dtoverlay=vc4-kms-v3d
root@plexamp4:~# cat /boot/config.txt |grep dtoverlay=hifiberry-
dtoverlay=hifiberry-digi
root@plexamp4:~# cat /boot/config.txt |grep force_eeprom_read=
root@plexamp4:~# cat /etc/asound.conf
cat: /etc/asound.conf: No such file or directory
root@plexamp4:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@plexamp4:~#'

Did you try my script with a clean/fresh install of the "Raspberry Pi OS Lite" image?

EgonHeuson, 2022-10-10:
Yes, I also tried to install `force_eeprom_read=0` but didn't work.
I used the guide you indicated with your link (as I mentioned above 😊)
Yup, I used your script over a fresh install. I also tried after with a manual install of plexamp but with your script to configure the HAT. I'm back home in 1h. I'll send you the config.txt file. :-)

odinb, 2022-10-10:
Well, then unless something weird is happening, you must have a bad card!

My full file (working):
`root@plexamp4:~# cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]

# --== Configuration for HiFi-Berry ==--
dtoverlay=hifiberry-digi
root@plexamp4:~#`

EgonHeuson, 2022-10-10:
Here is my config.txt :-) I commented the optional commands I tested

`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
#dtoverlay=vc4-kms-v3d,audio=off
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]

# --== Configuration for DIGI-DAC ==--
dtoverlay=hifiberry-digi
#force_eeprom_read=0
#dtdebug=1`

EgonHeuson, 2022-10-10:
With your config.txt, no more luck sadly :-( 

`root@RPiPlexamp:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

I'll try with an old RPi 3 I have when I have some time just to be sure.

EgonHeuson, 2022-10-11:
I confirm, it seems that the Pifi Digi+ is not compatible anymore with the latest Raspberry Pi 4. On an old raspberry Pi 2, Raspberry Pi OS 32B Lite, using only your script, it mounts perfectly as seen below:

[image]

[image]

Or maybe it is the 64B version of RPi OS that's in fault here. But I smell it more like an incompatibility with the RPi4. 
Hope this helps anyway :-)

odinb, 2022-10-11:
Well, I have 4 RPi4 where this is working with that same digi-card you have! I run 64-bit on all of them! So, no, it is not the RPi4!

EgonHeuson, 2022-10-11:
Is it the last version of RPi4 that you have? The one manufactured after April2022? Because it’s also this one that create an issue with the Hifiberry Digi2 Pro card.

odinb, 2022-10-11:
Ah, no! Mine are older! Bought just before prices went up! So, you are probably right then, it is the new hardware on the RPi4! Do you have a llink for this?

EgonHeuson, 2022-10-12:
> Do you have a llink for this?
You mean a link where I bought it? Or about this change in hardware? I found the info here: https://www.hifiberry.com/blog/digi2-pro-raspberry-pi-4-1-5-incompatibilities/
I didn't think about executing `cat /proc/device-tree/model`, but I'll try tonight. As I received mine in June, from a new production batch, I'm pretty sure that it is the v1.5. 
Also I have an older RPi4 on which I run [LibreELEC](https://libreelec.tv). I will perform some test on it to see if my assumptions are right. I'll keep you updated.

EgonHeuson, 2022-10-12:
Ok, my hypothesis is now fully confirmed. I have no problem with the Pifi Digi+ on my old RPi4. 
[image]
[image]

So I guess it would be important to indicate it to the community to avoid others to have such issue :-( Until Pifi proposes a new version of the card that is compatible with the RPi5 v1.5... :-D

odinb, 2022-10-12:
Ok, thanks for the information! Added this to the Q&A section!

EgonHeuson, 2022-10-13:
Thanks!! Too bad :-(