odinb/bash-plexamp-installer #8: node install on ubuntu jammy based systems

tgp-2, 2022-09-18:
Hello, just a heads up ... it seems the nodesource repo for node 12 is not supported on ubuntu jammy based systems. Jammy main repo still has node 12, so adding this as a fallback installer option might be useful. NVM is another option, but adding this might require more significant changes to the installer.

odinb, 2022-09-18:
Hi!

This installer is for a headless Plexamp running on RPi-hardware, so your Ubuntu-situation is not really an intended target of this project!

However, in order to be flexible, I added a switch for the Node.v12 install-part, making it optional!

Hope this works for you!

closed 2022-09-18 by odinb

tgp-2, 2022-09-18:
Thanks for the reply ... just to clarify, this would be for a headless install :)
With the raspberry pi supply shortage, many are looking for alternative devices ... in my case, I have an odroid-c4 and a nanopi-r4s ... both run ubuntu/armbian jammy (minimal/headless, not a full desktop), and both are great for Plexamp
So for this scenario, one could first install nodejs from the main repo, then run your installer and choose the option to skip node install ... think I got it, thanks!

odinb, 2022-09-18:
Do not have the Odroid-C4 myself, but I do have a couple of the Odroid N2+ (Running CoreElec on one, and Homeassisitant on the other).

Seems Armbian is available, assuming it is similar to Raspbian, so not Ubuntu.
https://www.armbian.com/odroid-c4/