odinb/bash-plexamp-installer #9: network-online.target for --user service

tgp-2, 2022-09-18:
Hello, one more item ... recent plexamp unit files from Plex require network-online.target to help with periodic headless device registration. As I understand it (and I'm no systemd expert), this target is visible/available for use with a system service ... which is what the Plex team sets up as part of their basic install instructions. However, I don't think the network-online.target is visible/available for use with a --user service. 

I don't fully understand all of the systemd dependencies, but when I use the unit file produced by this installer to set up Plexamp as a --user service, Plexamp fails to run after a reboot, and I get the following ...

```
odroid@odroid:~$ systemctl --user list-dependencies plexamp.service
plexamp.service
● ├─app.slice
● ├─basic.target
  │ │ └─...
● │ ├─paths.target
● │ ├─sockets.target
● │ │ ├─dirmngr.socket
● │ │ ├─gpg-agent-browser.socket
● │ │ ├─gpg-agent-extra.socket
● │ │ ├─gpg-agent-ssh.socket
● │ │ ├─gpg-agent.socket
● │ │ └─pk-debconf-helper.socket
● │ └─timers.target
○ └─network-online.target
odroid@odroid:~$ systemctl --user list-dependencies
default.target
● └─basic.target
×   ├─plexamp.service
●   ├─paths.target
●   ├─sockets.target
●   │ ├─dirmngr.socket
●   │ ├─gpg-agent-browser.socket
●   │ ├─gpg-agent-extra.socket
●   │ ├─gpg-agent-ssh.socket
●   │ ├─gpg-agent.socket
●   │ └─pk-debconf-helper.socket
●   └─timers.target
```
I think this means the plexamp service fails to run, and it appears that all other requirements are met except for the network-online.target. When set up this way, sometimes I've seen Plexamp start successfully, but not always ... and currently I can't get it to start at all.

Set up as a system service, Plexamp always starts successfully, and I get the following ...
(note, the unit file is same as what's produced by your installer, but with multi-user install target (same as in the original unit file from Plex)

```
odroid@odroid:~$ sudo systemctl list-dependencies plexamp.service
plexamp.service
● ├─-.mount
● ├─system.slice
● ├─network-online.target
● │ └─NetworkManager-wait-online.service
● └─sysinit.target
```
I think the solution for any users experiencing service startup issues is to disable the --user service and set up a system service using a modified unit file.

Sorry for such a long message, but I thought the output from the systemctl commands might be helpful.

odinb, 2022-09-19:
The answer is yes.
When running in user mode, the load path, by default, does not include system unit paths.

To accomodate your issue, I have modified the service-script to run from system, but start Plexamp as user.
This since it is never secure to run user-processes/services as an elevated user.

This should fix your issue, and still run the process as your user, i.e. still be safe!
To get the modification just run the installer again, choose correct user, choose "cleanup for upgrade", and "install or upgrade". Then reboot!

Since this script now cleans up the old user-service, and creates the new system-service, you will have to re-enable the service with "sudo systemctl enable plexamp.service". After this, it should not be needed for future upgrades!

closed 2022-09-19 by odinb

tgp-2, 2022-09-19:
Great, thanks so much for making the changes!