odinb/bash-plexamp-installer #13: .service file not replaced when reinstalling with another user

APMOadmin, 2023-01-30:


odinb, 2023-01-30:
You are not supposed to first install plexamp as one user, and then later install it as a different user. The script is not written that way. If you want to change the user after install, either perform cleanup of services etc. manually, or re-install the OS from scratch (faster most of the time) and re-run the plexamp installer as the correct user from the start.

APMOadmin, 2023-01-30:
thanks @odinb - manual cleanup is what I did and go it to work (well, install) after that. Just thought I mention it. The 'if' before creating the link to the service file s what tripped me. No biggie as we have the source code :)
Great work BTW - thanks a lot !!