odinb/bash-plexamp-installer #15: Cosmetic request - script should update the motd banner logo on subsequent releases

doasfan, 2023-02-04:
Hello,
My /etc/update-motd.d/20-logo displays the ASCII logo and "Plexamp-Linux-headless-v4.5.3" on ssh login. I've been hoping it would be replaced with an updated logo/banner/message of the day displaying the correct corresponding version, but I've done several upgrades and each version since has not updated the logo file.

Would be nice to have the logo updated with the correct version.

Thanks!

[image]

odinb, 2023-02-04:
Well, it actually does update, at least on all of mine!
This is currently line 281 of the script! (line number will change in future versions)

[image]

What OS are you running (output using "cat /etc/os-release")?

What do you see if you run the following 2 lines at command-line:
`PLEXAMPV="Plexamp-Linux-headless-v4.6.2"`
`sed -e "s#Plexamp-Linux-.*#"$PLEXAMPV\""#g" /etc/update-motd.d/20-logo`

You should get the motd with correct version, if not, let me know what you get!

odinb, 2023-02-08:
Since I have not heard back, I will close.

closed 2023-02-08 by odinb