odinb/bash-plexamp-installer #16: hdmi audio out on raspberry pi

tgp-2, 2023-02-25:
follow up to discussion on plex forum ...

on raspberry pi devices (3 and 4 for sure, don't know about others) + current/bullseye raspberry pi OS (or derivatives that use same kernel) ... hdmi audio output using vc4-hdmi alsa device doesn't work

pulseaudio can output audio to hdmi, but this requires Plexamp to be run as a --user service (system service can't talk to pulseaudio) ... --user service option would require some changes to the service file prepared by the installer, as described here ...
https://forums.plex.tv/t/plexamp-headless-on-a-rpi-3b-no-audio-out-over-hdmi/826971/4?u=tgp-2

another option ... the boot config file can be updated to select a different overlay for graphics driver/firmware, and after reboot there will be an hdmi audio device usable by alsa together with Plexamp run as a system service
https://forums.plex.tv/t/plexamp-headless-on-a-rpi-3b-no-audio-out-over-hdmi/826971/13?u=tgp-2

the pulseaudio + --user service option seems safer

odinb, 2023-02-28:
What exactly is the issue here? The script runs plexamp as a user-service already, which is your option 1 above.

tgp-2, 2023-02-28:
script uses system service (service file saved in or pointing to /lib/systemd/system) ... --user service file is saved in ~/.config/systemd/user (and has some different content vs system service file) and commands to enable/start/etc. use `systemctl --user` in place of `systemctl`

odinb, 2023-02-28:
But that is exactly how the script does it:

$ cat plexamp.service
[Unit]
Description=Plexamp
After=network-online.target
Requires=network-online.target

[Service]
Type=simple
User=pi
WorkingDirectory=/home/pi/plexamp
ExecStart=/usr/bin/node /home/pi/plexamp/js/index.js
Restart=on-failure

[Install]
WantedBy=multi-user.target
$

tgp-2, 2023-02-28:
that's a system service

tgp-2, 2023-02-28:
`systemctl status user.slice` shows list of running --user services
on a headless system, i think there's usually not much (if anything) running like this ... on a system running a desktop environment, there will be more user services (pulseaudio, etc.)

`systemctl status system.slice` shows list of running system services ... using the unit file prepared by the installer script and enabling/starting as instructed, plexamp will show up here

odinb, 2023-02-28:
Changed the dtoverlay-setting as part of the "Fix HDMI-audio setup".

tgp-2, 2023-02-28:
thanks, that's the method i'm currently using (for my one device using hdmi)

closed 2023-02-28 by odinb