odinb/bash-plexamp-installer #17: Request: Default to previously set timezone selection when the script is re-run

doasfan, 2023-08-04:
When using the script to upgrade an existing install, would be great if the timezone selection could simply default to whatever was it was previously set to, instead of requiring it to be selected again. Currently requires selecting the timezone, then confirming with option 5. Just clicking ENTER to accept previous set time would be better.

Example: I've installed it using option 4 for Pacific time. Upon upgrade, the script would ideally pull the previous selection and suggest by default to keep the last set timezone, with an option to change if desired.

odinb, 2023-08-04:
Hi! I know what you are saying, and could not find an easy fix for it.
However, you can just hit 5) to continue, no need to re-select the timezone! It will stay at whatever it currently is, so it is basically "defaulting" already!!
Hope this helps!

doasfan, 2023-08-04:
Option 5 "Quit" actually defaults to Central in the next prompt. Obviously
not a big deal, just being lazy and trying to save a couple keystrokes.

odinb, 2023-08-04:
Ok, will have a look at it, might take a couple of days!

odinb, 2023-08-04:
Done! Found a simple fix.

closed 2023-08-07 by odinb