odinb/bash-plexamp-installer #19: plexamp.service does not exist

Tecko71, 2023-09-12:
After runnigng the script it works , by creating the controller. but I can not enable and start plexamp.servce. Please help.

odinb, 2023-09-12:
What OS are you running on your RPi ("cat /etc/os-release" should give "Debian GNU/Linux 11 (bullseye)")? The script is only tested on this OS.

Did you run the script as root (it has to be)? What user did you choose during script execution (default is "pi")? This is the second prompt during script execution, and is the user you chose/added during creation of the pi-imager, unless you installed it manually later.

After installation, did you, as your user, run "sudo systemctl enable plexamp.service && sudo systemctl restart plexamp.service" to set plexamp to start on reboot, and to run it immediately? This user has to be the same as chosen during script execution.

What is the current user you want to use for the plexamp service (to get current user you can run "whoami")? If it is the "pi" user, defaults should work. If not, please change to the correct one at second prompt during script execution.

What does "sudo systemctl status plexamp.service" return?

Will leave this open for a day or two, if no response, I will assume it is fixed, and close.

Tecko71, 2023-09-12:
The problem is that the plexamp.service can not be found because I have to edit those. I did this:
[Service]
Type=simple
User=root
WorkingDirectory=/home/dietpi/plexamp
ExecStart=/usr/bin/nodejs /home/dietpi/plexamp/js/index.js
Restart=on-failure
Group=dietpi

I am running dietpi on a x86-64 PC.

odinb, 2023-09-13:
Try the script now and let me know outcome/error. Hopefully it is fixed!

Tecko71, 2023-09-13:
when I run the srcrip it also give this error, do not now how to fix that,. I think it means that the device is allready  registered:
root@dietpi:/home/dietpi# node /home/dietpi/plexamp/js/index.js
This browser doesn't support requestAnimationFrame. Make sure that you load a polyfill in older browsers. https://reactjs.org/link/react-polyfills
This browser doesn't support cancelAnimationFrame. Make sure that you load a polyfill in older browsers. https://reactjs.org/link/react-polyfills
Starting Plexamp 4.8.3
METRICS: Creating controller.

Tecko71, 2023-09-13:
the script is working non. I can use now sudo systemctl enable plexamp.service && sudo systemctl restart plexamp.service.

thx

odinb, 2023-09-13:
The "This browser doesn't support requestAnimationFrame." I think you can ignore, I got those also, and it still worked!

The second error is more troublesome, I did not get that one, and the service auto-starts on mine.

The third error I get also, **this is not how you start the application**, so **not an issue**.
dietpi@DietPi:~$ ls -all /home/dietpi/plexamp/plexamp/js/index.js
ls: cannot access '/home/dietpi/plexamp/plexamp/js/index.js': No such file or directory

So, for the second error, your path comes up incorrect in the service printout, one should be a softlink, the other the actual link as you can see from below printout. Did you fiddle with the service file on your own? If you did, please cleanup/remove the service-file(s) for Plexamp, as the script will re-create them, even on upgrade. So, just re-run script as an upgrade after cleanup to fix. Paths should be like this:
dietpi@DietPi:~$ ls -all /home/dietpi/plexamp/plexamp.service
-rw-r--r-- 1 dietpi dietpi 286 Sep 12 21:34 /home/dietpi/plexamp/plexamp.service
dietpi@DietPi:~$ ls -all /etc/systemd/system/plexamp.service
lrwxrwxrwx 1 root root 36 Sep 12 21:12 /etc/systemd/system/plexamp.service -> /home/dietpi/plexamp/plexamp.service
dietpi@DietPi:~$

Secondly, what version of NodeJS do you have? You can check like this:
dietpi@DietPi:~$ node -v ; npm -v
v16.20.2
8.19.4

Let me know what you find!

Tecko71, 2023-09-13:
I did fixed it ... needed to install node v16 with nvm, that directory is in this location ~/.nvm/versions/node/v16.20.2/bin/node

odinb, 2023-09-13:
As I suspected! This is most likely because you do not run the RPi version of dietPi.

But you still need to fix the service-files, as stated, one should be a softlink, the other the actual link.

Tecko71, 2023-09-13:
service-files?? That I do not understand which files you mean?

odinb, 2023-09-13:
As you can see here from your printouts:
sudo systemctl status plexamp
× plexamp.service - Plexamp
Loaded: loaded (/etc/systemd/system/plexamp.service; enabled; preset: enabled)
Active: failed (Result: exit-code) since Wed 2023-09-13 08:51:13 CEST; 8s ago
Duration: 885us
Main PID: 2399 (code=exited, status=203/EXEC)
CPU: 844us

It is pointing to a physical file "/etc/systemd/system/plexamp.service".

My script does not create/use that file as you can see from my printout:
dietpi@DietPi:~$ sudo systemctl status plexamp.service
● plexamp.service - Plexamp
     Loaded: loaded (/home/dietpi/plexamp/plexamp.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-09-13 08:17:12 CDT; 1h 42min ago
   Main PID: 432 (node)
      Tasks: 18 (limit: 2197)
        CPU: 7.498s
     CGroup: /system.slice/plexamp.service
             └─432 /usr/bin/node /home/dietpi/plexamp/js/index.js

Sep 13 08:17:12 DietPi systemd[1]: Started Plexamp.
Sep 13 08:17:14 DietPi node[432]: This browser doesn't support requestAnimationFrame. Make sure that you load a polyfill in older browsers. https://reactjs.org/link/react-polyfills
Sep 13 08:17:14 DietPi node[432]: This browser doesn't support cancelAnimationFrame. Make sure that you load a polyfill in older browsers. https://reactjs.org/link/react-polyfills
Sep 13 08:17:15 DietPi node[432]: Starting Plexamp 4.8.3
Sep 13 08:17:15 DietPi node[432]: METRICS: Creating controller.

The way it works is that the file "/home/dietpi/plexamp/plexamp.service" is the actual file as you can se here:
dietpi@DietPi:~$ ls -all /home/dietpi/plexamp/plexamp.service
-rw-r--r-- 1 dietpi dietpi 286 Sep 12 21:34 /home/dietpi/plexamp/plexamp.service

And the second file "/etc/systemd/system/plexamp.service" is actually just a softlink pointing to the actual file:
dietpi@DietPi:~$ ls -all /etc/systemd/system/plexamp.service
lrwxrwxrwx 1 root root 36 Sep 12 21:12 /etc/systemd/system/plexamp.service -> /home/dietpi/plexamp/plexamp.service

This is done by the script, so to fix, you can remove the file "/etc/systemd/system/plexamp.service", and then re-run the script as if you were updating, and the softlink will be created.

Tecko71, 2023-09-13:
when I remove /etc/systemd/system/plexamp.service than the next errors:

 systemd[1]: /etc/systemd/system/plexamp.service:10: Neither a valid executable name nor an absolute path: ~/.nvm/versions/node/v16.20.2/bin/node
Sep 13 18:25:16 plexampPi systemd[1]: plexamp.service: Unit configuration has fatal error, unit will not be started.
Sep 13 18:25:20 plexampPi systemd[1]: /etc/systemd/system/plexamp.service:10: Neither a valid executable name nor an absolute path: ~/.nvm/versions/node/v16.20.2/bin/node
Sep 13 18:25:20 plexampPi systemd[1]: plexamp.service: Unit configuration has fatal error, unit will not be started.

Tecko71, 2023-09-13:
thanks it should work now, no errors in systemctl status plexamp. Only that requestAnimationFrame.

closed 2023-09-13 by odinb