odinb/bash-plexamp-installer #20: Support for multi device streaming with SnapJack

RAYs3T, 2023-10-02:
Hey,

with the "original" installer, we also provision snapjack to enable multi-room / multi-device streaming.

This would be a great addition for this installer.

See: https://github.com/elan/snapjack-installer/blob/main/installer.sh

odinb, 2023-10-03:
That URL gives me a 404, can you please provide a functional link?

RAYs3T, 2023-10-03:
https://gist.github.com/RAYs3T/b1e0e07e5d3572459c06afbd8ffd8bbc

odinb, 2023-10-09:
What version of Plexamp will this work for? Plexamp-Linux-headless-v4.8.x and newer?
Looking into this now...

RAYs3T, 2023-10-09:
Good question, it's supported for at least the last 6 months, maybe even longer. Not sure what exact version.
I were using the beta releases, but this has already made it into the release.

RAYs3T, 2023-10-16:
Hey @odinb awesome - I can test it today as I have 2 PIs set up with it.
As far as I know it detects if SnapJack is preset and if there are multiple devices in the network.

elan, 2023-10-18:
Totally unreleased and unsupported. Don't make us feature flag it until release 😆

odinb, 2023-10-18:
> Totally unreleased and unsupported. Don't make us feature flag it until release 😆

Is it enough to have a disclaimer (see script), or you want me to fully remove?
Also, is it done correctly, os I am prepared for later when released?

elan, 2023-10-18:
Please remove until we release.

odinb, 2023-10-18:
Ok, done! Was the script correct install-wise otherwise for SnapJack?

elan, 2023-10-19:
SnapJack will have its own installer; from the outside in, so to speak.

closed 2023-10-27 by odinb