odinb/bash-plexamp-installer #22: Under Debian 12 (bookworm), audio HAT is not detected.

odinb, 2023-10-16:
Under Debian 12 (bookworm), the audio-HAT is not detected.

Debian 11 (bullseye) output:
root@PlexAmp:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@PlexAmp:~#

Debian 12 (bookworm):
root@PlexAmp:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@PlexAmp:~#

If you have a solution, please let me know.

tgp-2, 2023-10-18:
might only be some HATs not detected on bookworm ... I have an allo-digione, this one is recognized
```
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
card 1: sndallodigione [snd_allo_digione], device 0: Allo DigiOne HiFi wm8804-spdif-0 [Allo DigiOne HiFi wm8804-spdif-0]
```

odinb, 2023-10-18:
> might only be some HATs not detected on bookworm ... I have an allo-digione, this one is recognized
> 

Good to know, have a theory on why it is not working, hopefully will have time later in the week to fix.

Does the updated installer work for you on Bookworm with your card?

tgp-2, 2023-10-18:
strange ... i disabled the allo HAT and then tried to re-enable it using the installer, now it's not working (and i can't re-enable manually either)

odinb, 2023-10-18:
Seems my installer is removing the softlink "/boot/config.txt" pointing to "/boot/firmware/config.txt". Need to see how to best fix that.

odinb, 2023-10-18:
Softlink issue resolved.

closed 2023-10-18 by odinb