odinb/bash-plexamp-installer #24: Question: Help with other sound drivers / DAC HATs

newelement, 2023-12-21:
I'm looking to use Ian Canada's ESS DAC found here:

[https://github.com/iancanada/DocumentDownload/blob/master/RPiDacHAT/ES9038Q2MDualMonoDacHAT/ES9038Q2MPiDualMonoIIManual.pdf](https://github.com/iancanada/DocumentDownload/blob/master/RPiDacHAT/ES9038Q2MDualMonoDacHAT/ES9038Q2MPiDualMonoIIManual.pdf)

The docs state that the Audiophonics driver will work which is found here:

[https://github.com/audiophonics/I-Sabre_9038Q2M](https://github.com/audiophonics/I-Sabre_9038Q2M)

This is probably a bigger topic, but can you provide any help on getting Plexamp to see the driver?

newelement, 2023-12-21:
Is this where enabling I2S device comes in to play?

odinb, 2023-12-22:
According to this page: https://www.audiophonics.fr/en/dac-and-interface-modules/ian-canada-dual-mono-mkii-dac-es9038q2m-hat-raspberry-pi-i2s-spdif-pcm-dsd-p-18359.html

All that is needed is to:

newelement, 2023-12-22:
Oh sweet how did I miss that part. I have the HAT coming next week. I'll try it. Thanks.

newelement, 2023-12-22:
With `dtoverlay -a` I'm not seeing that overlay available. Is there another way to check that you know of ?

odinb, 2023-12-22:
Ok, added option for this card in the script. Let me know if it works!

john-pienaar, 2023-12-28:
Just to tack on to the above- it would be handy to support the newer Allo Boss 2 DAC:
`DIGICARD="dtoverlay=allo-boss2-dac-audio";;`
Also, thanks for this script!

odinb, 2023-12-28:
Ok, added the BOSS2 DAC, let me know of any issues!

newelement, 2023-12-28:
Reporting back - i-sabre-q2m is working great. Thanks

closed 2023-12-30 by odinb