odinb/bash-plexamp-installer #26: Help picking hardware

aunefyren, 2024-01-25:
Hey, thank you so much for creating this project and being so detailed in your documentation. I have never done anything with Raspberry Pi's, but I want to give this a go. Are you able to give me some tips? If not, feel free to close the issue. I figured this was the most appropriate place to ask for assistance,

I want optical audio, so I tweaked your hardware accordingly.

I have created a shopping list:
* Raspberry Pi 5 Model B 8 GB
* Official Raspberry Pi 5 USB-C Power Supply – EU – 5V 5A
* Sandisk Extreme Plus Micro-SD-card64 GB
* HIFI DiGi+ Digital Sound Card I2S SPDIF Optical Fiber with case (https://www.aliexpress.com/item/32998739264.html)

I am also considering this sound card: https://www.hifiberry.com/shop/boards/hifiberry-digi2-pro/

I have a few questions though;
* Do all my parts fit together?
* Am I missing anything I need for this to function?
* There is very little information on the AliExpress page and the product is very cheap. Do you think it supports 24bit 48kHz audio? 
* Should I spring for the HIFIBerry card instead? Would it be wiser? When I read the documentation it seems you bought from AliExpress.
* I cannot solder, but I have someone who is experienced with this. What information to they need to know how to connect the card to the Raspberry Pi? A diagram might be provided?
* Does one purchase cables to solder the cards together?

Really hope you can help me! 😁

odinb, 2024-01-25:
Hi! See responses below!//Odin

Do all my parts fit together?No, the case is for a different version of the RPi, you need a different case for the RPi5. For theRPi5, you might want the optional cooler/fan. Also, the RPi4 is more than enough if you intend it for Plexamp only, and it is cheaper! I run on both 2GB and 4GB versions of the RPi4 for all my Plexamp setups.
Am I missing anything I need for this to function?As stated above, case will not fit. Also, you might need the optical cable. The RPi4 needs a 5V3A PSU.
There is very little information on the AliExpress page and the product is very cheap. Do you think it supports 24bit 48kHz audio?Yes, that is the same clone I use.
Should I spring for the HIFIBerry card instead? Would it be wiser? When I read the documentation it seems you bought from AliExpress.Up to you, the clone works just fine.
I cannot solder, but I have someone who is experienced with this. What information to they need to know how to connect the card to the Raspberry Pi? A diagram might be provided?No soldering needed, the card is seated on the pin-header.
Does one purchase cables to solder the cards together?As stated above, card is connecting to pin-header, no soldering needed.

aunefyren, 2024-01-26:
Thank you so much @odinb, your answers were very insightful!

closed 2024-01-26 by aunefyren