odinb/bash-plexamp-installer #29: Alternate method for HDMI audio on RPi5

bhcompy, 2024-09-17:
I ran the script and couldn't get HDMI audio to work with either of the methods found in here or on the Plex forums (pulseaudio user service or changing the driver to fkms and  adding hdmi_drive=2 etc to config.txt).  Couldn't even detect a sound card (aplay -l returned no sound cards, and rapsi-config showed none, either).  Found some info stating that fkms is being deprecated and it sounds like some support was dropped from firmware in the RPi5, so I guess it makes sense.  The solution I found was to add the following line to /boot/firmware/cmdline.txt, reboot, and select the card in rapsi-config:

`snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0`

As far as the output, Plex sends it all as directplay and my receiver says it's receiving 7.1 (which is the max supported channels) but despite not showing it explicitly, it outputs whatever the source file is (so a 2ch/stereo track plays on the proper channels and a 6ch/5.1 surround track plays on the proper channels), and it picks up the proper sample rate and bitdepth.

n00b001, 2024-09-23:
Thanks for the fix - I've just encountered this issue myself.  

A week later I encountered this issue again (I did a apt full-upgrade and enabled overlayfs - "dtoverlay=vc4-kms-v3d" was reverted to "dtoverlay=vc4-fkms-v3d")

odinb, 2024-09-23:
> I ran the script and couldn't get HDMI audio to work with either of the methods found in here or on the Plex forums (pulseaudio user service or setting dtoverlay=vc4-fkms-v3d and adding hdmi_drive=2 etc to config.txt). Couldn't even detect a sound card (aplay -l returned no sound cards, and rapsi-config showed none, either). Found some info stating that fkms is being deprecated and it sounds like some support was dropped from firmware in the RPi5, so I guess it makes sense. The solution I found was to use dtoverlay=vc4-kms-v3d and dtparam=audio=on instead in config.txt, the following line to /boot/firmware/cmdline.txt, reboot, and select the card in rapsi-config:
> 
> `snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0`
> 
> As far as the output, Plex sends it all as directplay and my receiver says it's receiving 7.1 (which is the max supported channels) but despite not showing it explicitly, it outputs whatever the source file is (so a 2ch/stereo track plays on the proper channels and a 6ch/5.1 surround track plays on the proper channels), and it picks up the proper sample rate and bitdepth.

Since I am not using HDMI, but rather an audio-card, can you send me the references for this change? Also, please send me the exact changes you needed to do and the files they were done in. Is this change valid for RPi5 only, or all RPis? If all, I will see if I can fix the script, if not, it will be a comment in the wiki for now.

n00b001, 2024-09-23:
I recently migrated from a rpi4 to rpi5 - I have previously not had this issue with rpi4.

odinb, 2024-09-24:
Thanks for the additional input!
Updated the script, and changed all instances of "fkms" to "kms".
This should hopefully fix the issue, if not please let me know and what the issue/fix was/is!

Please keep in mind, this script is aimed at headless primarily, so if conflicts, the headless config will win.

Will leave the ticket open for about a week for you to reply, then I will close if nothing more is found.

bhcompy, 2024-09-25:
By all instances you mean on RPi5?  Or on every device?  I think fkms works on older devices (though maybe not preferred on the most recent OS, but it's not clear if these settings work on older RPi's as I do not have access to any to test)

odinb, 2024-09-25:
@bhcompy: Basically on any device! I tested on my RPi4, and it still worked after re-configuring the Audio-card. Did not try fresh install. Found this info, will put it in the wiki:

Here's what we know at time of writing about KMS (Kernel Mode Setting) and FKMS (Fake Kernel Mode Setting) on Raspberry Pi:

KMS vs FKMS
KMS (Kernel Mode Setting):
- Is the standard Linux framework for setting video modes
- Directly accesses hardware registers for HDMI/DSI initialization
- More extensible and open-source
- Generally preferred for newer Raspberry Pi models, especially Pi 4 or later

FKMS (Fake Kernel Mode Setting):
- A special driver for Raspberry Pi
- Uses the proprietary DispmanX API to set up HDMI
- Simpler for Linux to use but relies on closed-source components
- Was more common on older Pi models

Recommendations
For Raspberry Pi 4 or later:
- KMS is now the preferred and more actively developed option
- FKMS has some significant bugs on Pi 4 and is less supported

For older Raspberry Pi models:
- FKMS may still be a viable option, especially if having issues with KMS

Important Considerations
1. KMS on Pi affects more than just mode setting - it also configures 2D compositing planes through the DRM (Direct Render Manager) subsystem.
2. Switching between KMS and FKMS may require additional configuration changes:
   - Modifying boot/config.txt
   - Potentially adjusting hdmi_mode and group_mode parameters
   - Possibly editing /boot/cmdline.txt
3. The choice between KMS and FKMS can impact performance and compatibility with certain features or software.
4. If using the official Raspberry Pi 7" Touchscreen, additional overlays may be needed for KMS:
   dtoverlay=vc4-kms-dsi-7inch
   dtoverlay=rpi-ft5406

5. For the most up-to-date and model-specific recommendations, it's best to consult the official Raspberry Pi documentation or forums, as the preferred options can change with software updates.

bhcompy, 2024-09-25:
Oh fantastic.  Hopefully it works for everyone

bhcompy, 2024-09-26:
> Please keep in mind, this script is aimed at headless primarily, so if conflicts, the headless config will win.

I just wanted to clarify, the hdmi approach is still headless (video output is just the CLI, so it's useless), but it doesn't require purchasing a HAT (save $$) and it doesn't have the limitations of optical/spdif/rca (which are fine for 2ch audio, but have limitations for 6ch+ audio)

closed 2024-10-02 by odinb

odinb, 2025-07-30:
Thanks for adding!

Using KMS (as opposed to FKMS useful for older applications, hardware, or scenarios where the standard kernel driver does not work reliably—especially with displays that need nonstandard timings or have EDID issues) is more "mainline" and is the default as of Raspberry Pi OS Bullseye (and later Bookworm), and is recommended for the 5.15 kernel and newer.

So, basically if you want the most up-to-date graphics stack or are running on recent Pi hardware (Pi 4, Pi 5+), especially with modern Linux OS versions, KMS is preferred.

KMS relies completely on EDID. FKMS is more tolerant and often displays an image without a valid EDID. I would therefore most likely guess you have a problem reading the EDID information on your RPi4?

Things you could try are using a different HDMI cable or switching the HDMI port or monitor (if using).
Or are you by chance using the official Raspberry Pi 7" Touchscreen connected via DSI?

Let me know if you have additional information or findings.