odinb/bash-plexamp-installer #30: claim file will not persist

xanderphillips, 2025-01-29:
Ran script
install worked correctly
(pointed to user plexamp, which is a valid user)
ran node /home/plexamp/plexamp/js/index.js
got claim and pasted into cli
plexamp finishes installing
~/.local/share/Plexamp/Settings/%40Plexamp%3Auser%3Atoken exists

If I manually run plexamp via node cli command at this point it works 1 time, but stopping it erases the user token file

If I instead run node cli command and get token installed and token file generated then do a sudo systemctl start plexamp, it just erases the token file and fails to complete the startup of the service.

I've tried this with your script, as well as others, and with a docker container version, they all wipe out the token file when trying to start after it worked.
I've even tried on another fresh install setting the token file to read only immutable even by root but still able to be read to keep it from being erased, but then plexamp just fails to start altogether.  

At this point I'm at a loss, and wonder if it's a bug and not a faulty configuration since I've tried on so many fresh installs from so many different sources and the results are always the same with current OS and plexamp.

odinb, 2025-01-29:
Hi xanderphillips!

For your flow:

Ran script	> Odin: ok, good!
install worked correctly	> Odin: ok, good!
(pointed to user plexamp, which is a valid user)	> Odin: so, "plexamp" was the user the plexamp was installed to.
ran node /home/plexamp/plexamp/js/index.js	> Odin: Did you run this as the user "plexamp"? It has to be run as the user you installed to.
got claim and pasted into cli	> Odin: Did you run this as the user "plexamp"? It has to be run as the user you installed to.
plexamp finishes installing
~/.local/share/Plexamp/Settings/%40Plexamp%3Auser%3Atoken exists	> Odin: not sure, never gotten that on clean install, and I have installed hundreds of times.

If I manually run plexamp via node cli command at this point it works 1 time, but stopping it erases the user token file	> Odin: Did you run this as a service as instructed? Per the instructions at the end, and in the readme:
Now, again as your user, run sudo systemctl enable plexamp.service && sudo systemctl restart plexamp.service to set plexamp to start on reboot, and to run it immediately.

This means it should be run when logged in as the "plexamp" user in your case. After this, you can reboot, and plexamp should start normally.

If this still does not resolve it, please try on a clean/fresh Debian install, and I am pretty sure it will work! If you do not want to wipe your OS, grab a different SD-card to test.

Have fun!

xanderphillips, 2025-01-29:
I ended up just re-writing the service to run a script that first does
node /home/plexamp/plexamp/js/index.js then copies a backup copy of the
token back in place from the one that starting the service wipes out.  so
far that's working.  so weird that starting the service would delete the
existing token file instead of trying to use it.

odinb, 2025-01-29:
Seems a bit overkill, but sure, if it works!

closed 2025-01-29 by odinb